raspberrypi-ui/piclone #6: Cannot create file system

Cisecodev, 2016-06-03:
Raspberry Pi 2 - fresh rasbian install 2016-05-27 (not Noobs).
8GB SD card running Raspbian, another 8GB card (unformatted) in /dev/sda
Run SD Card Copier from menu
Result fat partition gets created and the I get the error Cannot create file system.
Repeatable.

Rpi-clone works fine.

spl237, 2016-06-03:
Check the destination card is not write-protected. Try formatting the destination card in another machine and try again. The error message is produced by parted being unable to perform an operation on the card; there is a problem with your card or card writer.

Cisecodev, 2016-06-03:
> Check the destination card is not write-protected
It is not write protected - I have tried several

> Try formatting the destination card in another machine and try again.
Done - no difference

> there is probably a problem with your card or card writer
tried three different makes not has an issue before.

As I said Rpi-clone works

Running Piclone from a terminal
[image]

Output of lsblk -o name,uuid
NAME        UUID
sda
└─sda1      9016-4EF8
mmcblk0
├─mmcblk0p1 2D2D-CD16
└─mmcblk0p2 2f840c69-cecb-4b10-87e4-01b9d28c231c

spl237, 2016-06-03:
OK, for some reason, /dev/sda1 on the target is not being unmounted when piclone asks it to; or actually looks as if it is being unmounted and then something else is remounting it before mkfs has time to do anything. Are you running any other software or processes that might be trying to access the card reader at the same time?

If you manually do a sudo umount /dev/sda1 before you start, does that help?

Cisecodev, 2016-06-03:
> Are you running any other software or processes that might be trying to access the card reader at the same time?

nothing else running - let alone anything that might access the SD card

> If you manually do a sudo umount /dev/sda1 before you start, does that help?
No difference, same error message

spl237, 2016-06-03:
It looks like something peculiar to your setup; I've tested piclone a lot and not seen a failure like this before. Does it happen with a completely clean install of Raspbian? What else have you installed?

Check the table of mounted devices - cat /proc/mounts, I think. Then force /dev/sda1 to unmount - sudo umount /dev/sda1 - and check the table again. Then run piclone - if it fails, check the table a third time - what is mounted in all three cases?

Cisecodev, 2016-06-03:
Tried brand new fresh install of Rasbian, installed xrdp - (I run headless) and ran Pi clone - it worked fine.

I just need to work out what I might have installed (I thought nothing) - or maybe a corrupt installation.

Closing this

closed 2016-06-03 by Cisecodev