raspberrypi-ui/piclone #7: [feature request] piclone on a loopback device (on a file created on a HDD)

golfromeo-fr, 2016-06-16:
it would be interesting to be able to use piclone on a loopback device

http://tldp.org/HOWTO/Cryptoloop-HOWTO/loopdevice-setup.html
=> instead of a real device, a file created on top of a filesystem

interesting for a board with a HDD attached (USB or SATA)

what is needed is to display the available /dev/loop? in the destination

Let's see what I can do to help here.

E3V3A, 2016-09-29:
[This](https://github.com/raspberrypi/piclone/pull/8) is a cool PR, why hasn't it been merged?

jtmoderate876, 2017-12-12:
I would really like to see a way to clone to an (reduced size) **image file** (on an attached USB, a NAS, etc. rather than just a hard copy overwriting a full sd or other drive). Perhaps this feature request is similar or accomplishes the same thing?