raspberrypi-ui/piclone #9: Allow piclone to write to SD card when running from external HD

mjculross, 2016-06-19:
It is possible (& very desirable) for RPi devices boot from SDcard, but run from a much larger external USB hard disk (e.g. WD PiDrive).  It would be nice to be able to directly copy the root partition from the (2nd partition on the) PiDrive back to the built-in SD card as a convenient backup, since, in this configuration, the RPi is not really running from the 2nd partition on the SDcard, but rather from the 2nd partition on the PiDrive.
With the current version of piclone, writing to the SD card is globally prohibited.  Rather than generically disallowing writing to any/all partitions on the built-in SDcard, would it be possible to disallow writing to the actual mounted devices/partitions instead, thus allowing the described backup ??

golfromeo-fr, 2016-06-19:
IMHO
http://support.wdc.com/KnowledgeBase/answer.aspx?ID=12968
"6. How to restore the latest factory image of the WD PiDrive OS onto the HDD
Technically, the Raspberry Pi currently can only boot from microSD™. However, you can put the bootloader on the microSD™ card and the rest of the OS on the WD PiDrive. Detailed instructions on how to do this are available on the WD Labs Community."

RPi can only boot from SD card.
while interesting "PiDrive" is just a SD card boot based with root filesystem on HDD

basic SD card reader reads/writes at 20MB/s 
USB-SATA is limited to RPi single USB hub... probably max 30MB/s

to come back to your question, piclone can be used to clone your SD card to the HDD drive
it will copy both boot/root partition, then you will need to change on the SD card ("boot.ini"?) to use the 2nd partition of the PiDrive HDD

PiDrive is a community-like NAS for RPi

As far as can see it has a USB3 interface while RPi has USB2 (a bottleneck)

mjculross, 2016-06-19:
Guillaume:
Thank you for you reply.  However, I must not have stated my question clearly.  I am already successfully running with the boot partition on the microSD & the root partition on the PiDrive.  I do not need to have piclone to copy the microSD to the PiDrive.  I've already accomplished that & am running that configuration.
What I want to do is to have piclone to copy the (actively used) root partition from the PiDrive to the (unused) root partition on the microSD to serve as a convenient (local) backup.  That way, if I make a change/update to the root partition on the PiDrive & mess something up, I can just restore the last backup of the root partition from the microSD to the root partition on PiDrive.
Right now, piclone is prohibited from writing to the microSD at all.  That restriction may be too tight, especially if (as in the case that I describe) the microSD is not being used except for booting !!  I can certainly do what I am asking for using dd, but piclone provides such a nice & convenient interface that allowing writes to the microSD would provide an easy way to do backups.
Maybe an "Advanced Mode" checkbox could be added that would remove the restriction from writing to the microSD and/or would allow copying individual partitions between devices.

Thanks again,

Mark J Culross, KD5RXT

golfromeo-fr, 2016-06-19:
@mjculross 

ok, your need is clear ;)

just add a partition on the SD card (you can use parted, gparted)

then use "rsync" from your HDD root file to the SD card second partition
http://linux.die.net/man/1/rsync

pay attention, your HDD partition is largely bigger than your SD card, so you can "rsync" only some folders.

IMHO piclone is not designed nor intended for your need (HDD=>SDcard) it is SDcard => SDcard mainly