raspberrypi-ui/piclone #11: The PARTUUID seems duplicate

david-foo, 2017-07-26:
We clone the SD card to a USB storage device using piclone.
What we want to do is to use SD card to keep the /boot partition and the USB device to keep the / patition.
We change the boot=/dev/sda2 in cmdline.txt to achieve this. 

We reboot the Pi for many times. Normally we see the /dev/mmcblk0p1 is mounted at /boot. But sometimes(about 20% rate) we see /dev/sda1 is mounted at /boot. 
It seems the PARTUUID is duplicate which caused the problerm.

So the piclone's work is: dd the Card-A to Card-B then rezise on Card-B. Am I right? 
Can we do it this way:
fdisk -> mkfs -> rsync

spl237, 2017-07-26:
Yes, under stretch, partitions with the same PARTUUID will not both mount. 

This is already fixed in the latest version of piclone - there is now a checkbox which allows you to force a new PARTUUID to be generated on the copy.

piclone already works without using dd, by the way - dd doesn't allow you to resize a card, and piclone is designed to enable this.

The new version will be released with the forthcoming stretch release of Raspbian, or you can download the latest code from this GitHub and rebuild it yourself.

closed 2017-07-26 by spl237

david-foo, 2017-07-26:
I guess there should not be dd because piclone is so fast ^_^. Thank you.