raspberrypi-ui/piclone #12: sudo' is no longer needed

david-foo, 2017-07-28:
When we start piclone on command line using "sudo piclone"it can not show the drop down list. So that the user can not choose the src device and the dst device. instead just run “piclone” can make it works.
But we can run “sudo piclone” properly with the former Raspbian image. Is this caused by the Raspbian or piclone itself?

spl237, 2017-07-28:
piclone uses a GIO volume monitor to detect drives. This requires access to a session DBus, which is available if the application is launched without sudo. However, if an application is launched with sudo, it doesn't have access to a session DBus and so cannot determine which drives are on the system.

The solution is to explicitly provide a session DBus - you do this by using "sudo dbus-launch piclone". This is how the new desktop file launches piclone, in order to get around this issue.

closed 2017-07-28 by spl237

christroutner, 2020-12-21:
I'm running Ubuntu v20 on the RPi and I wanted to make use of this piclone app. I managed to get it to install and I got stuck on this same issue. Luckily I knew enough to investigate the closed Issues to see if one covered my problem.

I'm unstuck, but how could people with this same problem find this solution easier?