raspberrypi-ui/piclone #13: Piclone image won't boot

dazz100, 2018-06-05:
Hi
I am having a problem with piclone.  I run piclone on a headless Raspi with:
`PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
`
The cloned image won't boot.
I don't have a monitor connected so I can't tell you what messages are being outputted.  I can see from the flashing green LED that it is failing early in the boot process.  

piclone is not working on multiple pi's running different software.  I have tried different SD cards.  I have tried doing full (not just quick) SD formats. 

I found a possible explanation here:  [https://raspberrypi.stackexchange.com/questions/68082/why-wont-my-raspberrypi-boot-if-i-use-parted-to-adjust-the-partition](url)
I don't know if this would affect piclone.

Let me know if I can provide additional info to help troubleshoot this problem.

Dazz

spl237, 2018-06-07:
I've never tried using piclone on a headless install, but given what it does in terms of messing about with partitions and the like, I'm not entirely surprised it doesn't work, as it may well become confused about which exact boot partition to copy. 

I think the answer for now is that piclone isn't supported on a headless image, but when I get some free time, I'll try and find out what is happening and see if it is fixable.

dazz100, 2018-06-08:
Hi

Not being able to clone SD cards is a major issue for me.  It takes 
about 2 days solid work for me to build a server installation from a 
blank SD card.

I though the only difference between the full Raspbian and the Lite was 
the selection of apps installed.

I suspect the problem with piclone may be the same as with pishrink

https://github.com/Drewsif/PiShrink/issues/24

I haven't tried pishrink yet but I will.

Regards

Darren Conway
36 Orr Crescent
Lower Hutt
New Zealand
ph +64  (0)4 569 1963

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

spl237, 2018-06-08:
I've just tried piclone on a headless Pi running over VNC, and the resulting image booted with no issues.

Setup was to take the latest released Raspbian with Desktop image (April 2018), flash to SD card, boot, enable VNC server, connect from VNC Viewer on Mac. Insert SD card in USB flash drive into Pi; run SD Card Copier from VNC Viewer and copy /dev/mmcblk0 to /dev/sda. Copy took 5-10 minutes as usual, showed "Copy complete" message at end. Shut down Pi, moved newly copied card into Pi SD card slot, restarted Pi - booted as normal and was able to reconnect from VNC Viewer.

Your comment about Raspbian vs Lite puzzles me, as I don't see the relevance to the issue under discussion here - piclone will not work on a Lite image, as it is a GUI application which requires a desktop to run.

I'm going to need more information about exactly what you are doing to be able to repeat the problem - as far as I can see, there is nothing fundamentally preventing piclone from working on a headless Pi.

spl237, 2018-06-19:
No updates on here - can you provide any information on how to repeat the problem you are seeing, or should I close the issue?

spl237, 2018-06-25:
Closing due to inactivity

closed 2018-06-25 by spl237