raspberrypi-ui/piclone #15: Two things needed to be installed before .autogen worked.

kwende, 2018-12-08:
uname -a
Linux fred 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

Using raspbian. I had to install libglib2.0-dev because of the following error when executing autogen: 

_configure.ac:152: error: possibly undefined macro: AM_GLIB_GNU_GETTEXT
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation._

I also needed to install intltool  because of: 

_./autogen.sh: 5: ./autogen.sh: intltoolize: not found_

I think the documentation should be updated.

spl237, 2018-12-08:
The build documentation assumes, as is standard for Debian packages, that all dependencies have already been installed - the easy way to do this is "sudo apt-get build-dep piclone".

closed 2018-12-08 by spl237

flatsiedatsie, 2019-06-09:
Had the same issue on a headless install.

`intltoolize: not found`

Could the script perhaps try to install these things if they are found to be missing?

flatsiedatsie, 2019-06-09:
Also:

```
pi@gateway:~/piclone $ sudo apt-get build-dep piclone
Reading package lists... Done
E: You must put some 'source' URIs in your sources.list
```

spl237, 2019-06-09:
See above - the build instructions assume you know the basics of building and installing a Debian package. It is your responsibility to set your system up for building by enabling download of source packages and installing dependencies. As above "sudo apt build-dep piclone" will automatically install all required dependencies, and this is a standard step before building any package.