raspberrypi-ui/piclone #16: Creating an error on the new card

npaisnel, 2019-01-16:
Well, I can only trust the person that linked me to here, that pic lone is the same app as the SD Card Copier that is on the RPi Desktop image...;  I have a problem with it and was told to report it here.

I have been trying to Clone a 64gb card,  with Rasp Desktop on it...it is my 'working GUI' with samba/ssh/ gparted etc etc ..things i want every day set up on it.

The actual size is only about 3gb.  I wanted an image file so i can burn it back to a new card with Etcher when ever I need it.

Other way sI know...but lets just tell you about the error I get

Copying 64gb card down to an 8gb card.

The copy on the 8Gb card, copies and works correctly to boot the Pi.....but ..if I then boot the Pi with another SD card and examine the 8Gb copy in GPARTED I get this...an orange Warning Triangle on the 8Gb card and this message:

```
Unable to Read the contents of this file system!
Because of this some operations may be unavailable
The Cause might be a missing software package
The following list of software packages is required for ext4 file system support" e2fsprogs v1.41+
```

https://www.raspberrypi.org/forums/download/file.php?id=27497

`pi@raspgui:~ $ sudo fdisk -l -u /dev/sda
Disk /dev/sda: 7.4 GiB, 7969177600 bytes, 15564800 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xfde85379

Device     Boot Start      End  Sectors  Size Id Type
/dev/sda1        8192    96663    88472 43.2M  e W95 FAT16 (LBA)
/dev/sda2       98304 15564798 15466495  7.4G 83 Linux`

The original 64gb card is fine...but the 8gb copy has this error, which when I come to use dc3dd to image the 8gb card, it either fails then, or the card written from the new image fails with “Error: "Can't have a partition outside the disk!”

Basically what I am trying to do is to create an image of the  64gb card ...but unfortunately I do not have a bigger card to boot the Pi from in order to create the img on to so I have to copy the 64gb card to a smaller card, then image that card.

Yes I could use GPARTED to shrink the FS..but as I have only this one copy of this Master OS SD Card, I don't want to fuck with it till I have cloned it.  It has taken months to set up this master card and I don't want to screw it up

ZanderBrown, 2019-01-17:
So what do you expect PiClone to do?
You may have < 8GB on the source card but as the partitioning is for 64GB the target is just too small

npaisnel, 2019-01-17:
Just do what the Help file suggests it will do.

My destination card does have ample space to hold the data on the 64 gb card at 3 gb.
I'd expect the SD Copier to do what ever is needed, by adjusting partitioning etc to make it work.  There is no warning in the help to imply that it will not work.
I did also try 
64-32
32-16
16-8

> Note also that the destination card doesn’t need to be the same size as the source card, but it must have enough space to hold all the data that is on it. The application will warn you if there is insufficient space on the destination.