raspberrypi-ui/piclone #18: cannot install piclone

huongminh, 2019-05-17:
I had an issue with the desktop sd card copier on my pi3b running stretch. When I put a brandnew sd card in the usd card reader and plug it in a usb slot, the card copier works well. But when I put in a card that has been cloned from my Pi before (like I want to rotate backup), I got an pop up message: Operation Cancelled. This error used to happen on and off but now it permanently happens every time I do the copy. I would have to format the card on my PC with the SD card formatter then put it back to the Pi, then it works. 
So I tried to re-install piclone here but came across this error. Any idea to fix both issues. many thanks!

pi@raspberrypi:~/piclone $ dpkg-buildpackage -us -uc
dpkg-buildpackage: info: source package piclone
dpkg-buildpackage: info: source version 0.13
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Simon Long <simon@raspberrypi.org>
Undefined subroutine &main::debarch_to_cpuattrs called at /usr/bin/dpkg-architec                 ture line 248.
dpkg-buildpackage: error: dpkg-architecture subprocess returned exit status 255

spl237, 2019-05-21:
Can't repeat this on my image - I suspect it may be something to do with the cards you are using.

I suspect the rebuild problem may be caused by missing dependencies - try "sudo apt build-dep piclone" before you start.

Askmewho, 2020-11-02:
@spl237 oka, just bc you won't let me answer... I do have several rpis. they are good ones, they teach me how bad broadcom hardware is so I turned to other hardware, with more I/O, linux friendly (no threadX),real GPUs, and no hypocritically based on charity and having his CEO working at broadcom. you all have the face as a rock. no charity, just a broadcom sbc.

closed 2020-11-02 by spl237