raspberrypi-ui/piclone #22: SD Card Copier for debian/Ubuntu ARM64 or ARMHF (64 bit)

tikiatsbme, 2019-10-30:
Hi, I also requested this elsewhere, but this may be a better forum

https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial - ISSUES "SD CARD COPIER"

in part "...We need a GUI way to easy make a clone of a working micro SD card. Not sure if "env", "Sudo=_ASKPASS", "Usr/lib/piclone/pwdpic.sh", "usr/bin/piclone" even exist in an Ubuntu server image or Ubuntu desktop image. I doubt that Ubuntu would incorporate such, so its needs to be added somehow. ...Just Wondering.........."

spl237, 2019-10-31:
My apologies, but we don't have the resource to provide support for other operating systems. Ubuntu or any other OS are welcome to pull in the tools we have created for Raspbian, but our development efforts will only be on Raspbian / Debian themselves.

closed 2019-10-31 by spl237

tikiatsbme, 2019-10-31:
Whether closed or not, your app is CRITICAL for any OS running on  a Raspberry Pi due to its unique boot off so card. Please at least email what files I need, where to put them and what each line in terminal will be. I,ll build it and post to Ubuntu and Here for all FOSS users to use as they wish. PLEASE!!!!!!!

XECDesign, 2019-10-31:
Everything you need is in this git repo already. No need for anybody to email anything.

spl237, 2019-10-31:
The SD Card Copier app is not "critical"; it does nothing magical or Pi-specific. 

It just creates the same layout of partitions on the new SD card as on the old one, and does straight file system copies of every file in every partition on the source to the equivalent partition on the target. The procedure can easily be deduced by reading the piclone.c file in this repo, as it does everything by making system calls to standard UNIX file system tools.

tikiatsbme, 2019-10-31:
The average GUI desktop user thinks it’s “ Critical”, just because us semi-geek, and super geeks think it’s nothing, doesn’t make less Critical for that user that Linux needs to have more of. It was a simple request, guest I am on my own..........

Thomas J. Killeen, PE, RMS
46 Woodland Hills
South Berwick, ME 03908-1834
207-752-4102
Thomas.killeen@icloud.com

procount, 2019-10-31:
> Everything you need is in this git repo already. No need for anybody to email anything.

Absolutely agree with this.
I used it as the basis for the Card Clone facility in PINN. It just needed converting from GTK to Qt.

XECDesign, 2019-10-31:
> It was a simple request, guest I am on my own..........

You're not on your own, all the work has already been done for you.

Somebody (Simon) took time to write this tool, create all the packaging needed to get into any Debian-based distro (including Ubuntu) and kindly put it up for anybody to use as they see fit.

You have all the files you could possibly need right in front of you.