raspberrypi-ui/piclone #24: Writing to wrong device!

Botspot, 2020-06-08:
See: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=276590
Basically, I selected one destination drive out of the list, but SD Card Copier wrote to a different one.
Years-worth of pictures have been lost. :(

ghollingworth, 2020-06-09:
Can you specify the steps you follow (exactly) to reproduce this behaviour?  Obviously it's not supposed to do this!  Also can you try it with different SD cards / drives etc to see if there's something particular about your cards that is the problem (i.e. whether the issue only occurs when you do it with your particular card / SD card reader / drive)

spl237, 2020-06-09:
Looking at the code, I cannot see how this can happen.

When a copy starts, the friendly name and device name of the target device is read from the combobox, and strtok is used to separate them at the brackets - so the variables dst (the friendly name) and dst_dev (the device name) are read at the same time and dst_dev is only ever written to at that point. The "are you sure" prompt uses dst, and thereafter the copy uses dst_dev.

Unless I am missing something - and the fact that this is the first report of this issue in four or five years suggests that this code is doing the right thing - I can't see how this can fail.

I wonder if the fact that you have so many different devices connected to your Pi is causing devices to change somehow, so that what was /dev/sde was the correct target when you were prompted, but it changed underneath you to a different physical device? Your mention of a DVD drive stands out - I'm not sure how device naming works with a device with removable media in the system, for example.

JamesH65, 2020-06-09:
What would happen if you ran up the copier, it filled its combo list, then you plugged in another device - would that change things enough underneath to break it? Or does the combobox get updated if the underlying devices change?

spl237, 2020-06-09:
There is a watch on the volume monitor as part of the program - if devices change, the combobox updates automatically.

However, if you plugged in or unplugged a device after clicking "start" while the confirmation dialog was on the screen, that could change devices underneath you and mean you wrote to the wrong device. But there's (hopefully) a relatively narrow window in which that could happen.

I could modify the code so that the confirm dialog is closed and you are kicked back to the main screen to reselect if a volume change is detected while that dialog is on screen, I guess.

Botspot, 2020-06-09:
> Can you specify the steps you follow (exactly) to reproduce this behaviour?

Unfortunately I don't know exactly what is causing this. I've begun running tests with varying configurations to see what I can narrow down.

> However, if you plugged in or unplugged a device after clicking "start" while the confirmation dialog was on the screen, that could change devices underneath you and mean you wrote to the wrong device.

I may vaguely remember a "device was removed before ejecting" notification around the same time. You may be onto something! (though with so many devices, I have no way of knowing which device was "removed".)

I'll keep running tests. So far I've not been able to reproduce it today, though I find it strange that the memory card activity light flashes when SD Card Copier creates its partitions on the SSD.

spl237, 2020-06-09:
> I may vaguely remember a "device was removed before ejecting" notification around the same time. You may be onto something! (though with so many devices, I have no way of knowing which device was "removed".)

That would seem to me to be a "smoking gun" - the only way I can see that what you describe can happen is a device disconnecting and devices moving around after you had clicked "Start" and while the confirmation screen was on display.

It's probably a sensible change for me to close that window and drop back with a "devices changed" message if the drives do change while it is shown.

spl237, 2020-06-16:
I've just released a new version of piclone which monitors for changed drives, and aborts a copy if a drive is added or removed at any point; hopefully this will prevent a recurrence of the problem.

It should be available in apt in the next day or so.

closed 2020-06-16 by spl237