raspberrypi-ui/piclone #25: [feature request] missing possibility not to use UI, but text parameters

bufemc, 2020-09-02:
I am stuck when using sudo nearly like https://github.com/raspberrypi-ui/piclone/issues/14
but with 

```
pi@raspberrypi:~ $ sudo piclone
No protocol specified

(piclone:1497): Gtk-WARNING **: 10:48:16.647: cannot open display: :10.0
```

I am connected via W10 remote desktop. I read about the su cannot access the x server that easily etc etc.
So what about the possibility instead of using the UI to just have an easy (optional) text in- (via parameters) and output?

piclone in=/dev/bla1 out=/dev/bla2

In the end I see myself falling back to dd ,-)

holta, 2020-09-02:
Consider https://github.com/georgejhunt/piclone_cmd created by @georgejhunt

For the http://Internet-in-a-Box.org project, where X (graphical desktop) is not always available in developing world schools.

(Based on https://github.com/nwright-mcc/piclone_cmd with support for aarch64 added.)

jvonau, 2020-09-02:
Wasn't piclone_cmd going to become part of the admin-console?

spl237, 2020-09-02:
Really not the place for this discussion - piclone is a GUI application; a command-line application would be a completely separate project.

closed 2020-09-02 by spl237

holta, 2020-09-02:
> Wasn't piclone_cmd going to become part of the admin-console?

It is.

FYI piclone_cmd (command-line version of piclone, described above) is now part of Internet-in-a-Box's Admin Console, as documented here:

http://FAQ.IIAB.IO > "44 How do I back up, shrink & copy IIAB microSD cards?"