raspberrypi-ui/piclone #27: failing on another arm sbc

Askmewho, 2020-11-02:
Hi! I wasnt expecting to having this working out of the box on another platform like rk3399 here, but.
the thing is the next one. if I run pi clone without sudo it does detect my drives but cant work as expected

pi@rockpi-4b:~$ piclone 
sh: 1: uuid: not found
WARNING: You are not superuser.  Watch out for permissions.
umount: /dev/mmcblk2p9: No such file or directory
umount: /dev/mmcblk2p8: No such file or directory
umount: /dev/mmcblk2p7: No such file or directory
umount: /dev/mmcblk2p6: No such file or directory
umount: /dev/mmcblk2p5: No such file or directory
umount: /dev/mmcblk2p4: No such file or directory
umount: /dev/mmcblk2p3: No such file or directory
umount: /dev/mmcblk2p2: No such file or directory
umount: /dev/mmcblk2p1: No such file or directory
1+0 records in
1+0 records out
512 bytes copied, 0.00233624 s, 219 kB/s
Error: Partition(s) 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64 on /dev/mmcblk2 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.

if I do it with sudo it doesnt detect any device.
how we could make a workaround for this or debug it??
thanks!!

spl237, 2020-11-02:
piclone requires underlying software components from Raspbian, which if you are running on another SBC will not be present. We don't support running it on non-Pi hardware.

closed 2020-11-02 by spl237

Askmewho, 2020-11-02:
what component? @spl237

Askmewho, 2020-11-02:
if you respond so unrespectfully at lest give details

spl237, 2020-11-02:
The person displaying a lack of respect here is you, by expecting us, a charity, to expend time (and hence money) supporting competitor products. If you want support, buy a Pi rather than a product from one of our competitors, and support our charitable activities in the process.