raspberrypi-ui/piclone #28: [Insuficient space. Backup aborted] when cloning from 32Gb SD car to 256Gb SD card.

cmarchand, 2020-12-01:
Hello,
my pi is booted from a 32Gb SD card, and I want to move the whole install to a 256Gb SD Card. I expect this should be Ok.

But, in UI, I get this message : `Insuficient space. Backup aborted.`

There is no message into the console (I've started whith `sudo dbus-launch piclone`).

Source card partitions and target card partitions are different :

### Source
```
pi@max2play:~ $ sudo fdisk /dev/mmcblk0

Welcome to fdisk (util-linux 2.29.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): p
Disk /dev/mmcblk0: 29,7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xee397c53

Device         Boot    Start      End  Sectors   Size Id Type
/dev/mmcblk0p1          8192    93813    85622  41,8M  c W95 FAT32 (LBA)
/dev/mmcblk0p2         94208 61933952 61839745  29,5G 83 Linux
/dev/mmcblk0p3      61933953 62333951   399999 195,3M 83 Linux
```

### Target card
```
pi@max2play:~ $ sudo fdisk /dev/sda

Welcome to fdisk (util-linux 2.29.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): p
Disk /dev/sda: 238,3 GiB, 255869321216 bytes, 499744768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xee397c53

Device     Boot    Start       End   Sectors   Size Id Type
/dev/sda1           8192     93813     85622  41,8M  e W95 FAT16 (LBA)
/dev/sda2          94208  61933952  61839745  29,5G 83 Linux
/dev/sda3       61933953 499744767 437810815 208,8G 83 Linux
```

Source boot partition is FAT32, but target boot partition is FAT16.

I use Card Copier V1.0

Any help will be much appreciated

spl237, 2020-12-01:
Hmm. Did you happen to see which partition caused the problem? If you look at the dialog while the copy is running, it will say "copying partition n of m", where in your case m should be 3. I've had a quick look at the code and I can't see anything amiss; my first thought was that the size read from the 256 Gbyte partition was so large that it was being truncated in a variable somewhere, but I can't see how that can have happened.

To double check, can you please insert both source and target cards and paste the output of "df" here?

cmarchand, 2020-12-01:
```
pi@max2play:~ $ df
Sys. de fichiers blocs de 1K  Utilisé Disponible Uti% Monté sur
/dev/root           30419692 28926184     229012 100% /
devtmpfs              443152        0     443152   0% /dev
tmpfs                 447760    22876     424884   6% /dev/shm
tmpfs                 447760    47876     399884  11% /run
tmpfs                   5120        4       5116   1% /run/lock
tmpfs                 447760        0     447760   0% /sys/fs/cgroup
/dev/mmcblk0p1         42136    22857      19279  55% /boot
tmpfs                  89552       12      89540   1% /run/user/1000
/dev/sda2           30303364    45080   28695908   1% /media/usb0
/dev/sda1              42708    23014      19694  54% /media/usb1
```
It fails at `2 of 3`.

cmarchand, 2020-12-01:
Sorry, it's in french, but I guess you'll be able to translate column headers...

cmarchand, 2020-12-01:
Console displays this :
```pi@max2play:~ $ sudo dbus-launch piclone
umount: /dev/sda9: mountpoint not found
umount: /dev/sda8: mountpoint not found
umount: /dev/sda7: mountpoint not found
umount: /dev/sda6: mountpoint not found
umount: /dev/sda5: mountpoint not found
umount: /dev/sda4: mountpoint not found
umount: /dev/sda3: not mounted
1+0 enregistrements lus
1+0 enregistrements écrits
512 bytes copied, 0,0067323 s, 76,1 kB/s
fatlabel: warning - lowercase labels might not work properly with DOS or Windows
mke2fs 1.43.4 (31-Jan-2017)
Re-reading the partition table failed.: Périphérique ou ressource occupé
Warning: The resulting partition is not properly aligned for best performance.
```

cmarchand, 2020-12-01:
And source card is the on used to boot pi

spl237, 2020-12-01:
OK, 2 of 3 means it's not the big partition, so it's not a variable overflow.

I'm guessing you only have the one 256GB card, so don't have another to try?

I'll need to find a similar sized card and repeat the test here - give me a few days to see if I can get my hands on one. To be honest, I've never used the copier with a card that big, so there may be something about big cards which is causing an issue.

spl237, 2020-12-01:
One thing does come to mind - are you sure that it is a genuine 256GB card and not a clone card which claims to be a lot bigger than it really is? A lot of cheap cards sold on eBay and the like as 256GB are actually only 32GB or smaller. It doesn't look like that is the case here, but just to be sure - have you tried using the card in a camera or PC and copying a lot of data to it?

cmarchand, 2020-12-01:
Well, I didn't try. It is a SanDisk Ultra 256Gb, bought in a famous store in france, Fnac. Sure, it may be a fake, but i do not this so.
I'm gonna try tomorrow, but it'll take time !
[image]

spl237, 2020-12-01:
OK, if it's a SanDisk-branded card from a reputable retailer, it's probably a genuine card, so don't worry about testing it; I'll get hold of one here and do some tests to see if I can repeat the problem.

cmarchand, 2020-12-02:
I've tested, and I have now 20 files for 70Gb on the card, formatted with only one partition, JHFS+ type. So the card is able to store more than 64Gb.

spl237, 2020-12-02:
OK, thanks for checking - I'm going to order a couple of 256GB SanDIsk cards myself and see if I can see what is going on. They might not be here until next week, so please bear with me until then!

cmarchand, 2020-12-02:
Perfect for me, no emergency...

spl237, 2020-12-06:
OK, I bought a couple of exactly the same SanDisk 256GB cards, and I've just cloned my current 16GB card onto one of them with no problems - the copy went fine with no errors, and I can boot the Pi off the 256GB target card and see all the free space. So there is nothing fundamentally broken in piclone with regards copying to big cards; it must be something in your image.

You seem to have three partitions. A standard Pi card only has two - are your cards created by NOOBS? If not, where is the third partition from?

Have you tried copying to any other cards, such as another 32GB card? Do you get the same error?

cmarchand, 2020-12-07:
The system is a Max2Play, installed from [downloadable distro](https://www.max2play.com/en/max2play-image/), which is based on a Raspbian.

My `uname -a` gives this :
```bash
Linux max2play 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

I have no idea where this third partition comes from. As it is very small, it should contain nothing... and as it is not mounted, in fact, it's not used ! Probably an aligned partition created to fill remaining space...

spl237, 2020-12-07:
OK, that site has both Stretch and Buster images - which are you using? I'll download and have a look.

cmarchand, 2020-12-07:
How can I check this ?

spl237, 2020-12-07:
lsb_release -a from a terminal window will tell you.

cmarchand, 2020-12-07:
```bash
pi@max2play:~ $ cat /etc/issue
Raspbian GNU/Linux 9 \n \l
pi@max2play:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@max2play:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.11 (stretch)
Release:	9.11
Codename:	stretch
```

So, Stretch !

spl237, 2020-12-07:
I'm downloading the stretch image now to check, but I suspect the problem is that it will be using a very old version of piclone which may well not have properly supported large cards. An image based on stretch will be almost two years old by now - there have been a lot of changes to piclone since then. I'll check...

cmarchand, 2020-12-07:
So, I clone the repo, `make configure`, `make install` and so on ?

spl237, 2020-12-07:
Let me check first - I'm just flashing the image now.

spl237, 2020-12-07:
OK, it's definitely an old version in that image - 0.12 vs 0.16 - so I think the first step in any attempt to find out what is going wrong is to be to be on the latest version. So yes, you'll need to pull the image from this repo; there are build instructions in the README.

I should say that we don't in general provide support for OSes other than Raspberry Pi OS, nor for anything other than the current release of that (i.e. buster), so if an update to the latest version of piclone doesn't fix it, I probably won't be able to provide much help; I don't know what max2play might have done to their image!

cmarchand, 2020-12-07:
Ok, I'll do this tonight, and I'll keep you informed.

cmarchand, 2020-12-07:
Do dependencies have change between 0.12 and 0.16 ?

spl237, 2020-12-07:
No idea - "sudo apt build-dep piclone" will install all dependencies.

cmarchand, 2020-12-07:
```bash
configure: error: Package requirements ( gtk+-2.0 >= 2.18.0) were not met:

No package 'gtk+-2.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables PACKAGE_CFLAGS
and PACKAGE_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

pi@max2play:~/piclone $ sudo apt build-dep piclone
Lecture des listes de paquets... Fait
E: Vous devez insérer quelques adresses « sources » dans votre sources.list
```

Last line translated: you have to insert few sources addresses in your sources.list

spl237, 2020-12-07:
OK, sorry - I really cannot spare the time to walk you through the process of building a Debian package. There are lots of online resources where you can look this up, such as https://wiki.debian.org/BuildingTutorial

cmarchand, 2020-12-07:
I've managed to build and install piclone from git repo. But without success in cloning.

Consider this issue as a too old version issue, or a bad luck issue.

Thanks a lot for the time you have spent on this.

Is there a way to support / sponsor your work ?

Best regards,
Christophe

closed 2020-12-07 by cmarchand

spl237, 2020-12-07:
> Is there a way to support / sponsor your work ?

Just buy a Pi, and you've already done that!

Sorry you couldn't get it working - when I ran the original version of Piclone in the max2play image I downloaded, it didn't throw any insufficient space errors, and copied the first two partitions fine, but failed to mount the third partition to copy that. So it wasn't working, but in a different fashion from the way it failed for you, which makes me suspect there is something odd about the max2play images.