raspberrypi-ui/piclone #29: [Suggestion] Add a feature to allow the creation of image file snapshot

SMH17, 2020-12-08:
The requirement of an external USB card reader to create a clone backup isn't very handy. 
It would be much better including a feature to allow also the creation of flashable image files so that you can save the snapshot backup images where you want (and possibly even schedule them).

spl237, 2020-12-08:
While I agree that this would be useful, it is such a fundamental change to how piclone operates that it really isn't practical to modify it to do this. Other Unix command line tools, such as dd, already exist to do this.

closed 2020-12-08 by spl237

SMH17, 2020-12-08:
@spl237  To achieve this, You could consider to route the operation through a loopback \ virtual drive if you want to avoid radical changes to the current implementation.