raspberrypi-ui/piclone #35: Creating executable desktop shortcut for piclone?

RobotsAreCrazy, 2021-03-07:
Hi, 

Somehow managed to make the piclone app for aarch64 it sits in /usr/bin/
how could i add a shortcut that will run it as doesn't launch like a normal app i have to use application finder on the shortcut  then that opens the piclone in bin folder

not tested it yet, but as a handy linux novice i see actually getting it building it as a acomplishment i just want to know how to have it launch directly from a shortcut on desktop as how do i make it avavailable to all users?

Thankyou for creating the git for it as didn't want to use pi os as 64Bit it is not ready, so using ubuntu 20.04.2 with (xubuntu de)  probably switching to MATE de soon on a new install so good to know how to do things.

Thanks again,
Baz

spl237, 2021-03-07:
Sorry, but I don't know how desktop shortcuts work on Ubuntu - you're going to need to ask someone who does! The command line used to launch piclone is in the piclone.desktop file in the data subdirectory of this repo - you'll need to put that into whatever mechanism Ubuntu uses for shortcuts.

closed 2021-03-07 by spl237

RobotsAreCrazy, 2021-03-07:
Hi Simon, is there an os for the pi4 you would recommend?

spl237, 2021-03-07:
The only one I recommend is ours - the benefits of a 64-bit OS are very minimal and not really worth having for 99% of users (the vast majority of Linux apps don't need to access that much memory anyway), so the fact that RPiOS is still 32-bit isn't really a reason not to use it for most people. Given the amount of hardware optimisation in RPIOS, it will easily outperform any other 64-bit OS.

RobotsAreCrazy, 2021-03-07:
I have a 8GB pi4 are you still recommending pi os 32bit?

spl237, 2021-03-07:
Yes, absolutely. 

Very few apps use anywhere near 8GB of memory - other than some heavy-duty database applications which you are very unlikely to be running, most apps use nowhere near the 4GB that can be accessed with a 32-bit OS. Further, the 4GB limit is per-application; the real advantage of an 8GB Pi is that multiple applications can run simultaneously and each of them can access up to 4GB - so you can have two separate applications each of which can use 3GB, for example, and will still have 2GB free.

Most people don't get any benefit at all from a 64-bit OS - in fact, a lot of 64-bit code will run *slower* than the equivalent 32-bit code, because the larger addresses mean the code is bigger, takes longer to load etc. A bigger number does not automatically mean something is better or faster!

RobotsAreCrazy, 2021-03-07:
Well, i do have further plans...

portainer + docker + unify controller + adguard home + nextcloud

so still 32Bit best in that use case?

spl237, 2021-03-07:
Yes - nothing there is likely to require more than 4GB per application. And as before - other OSes do not have the numerous custom hardware optimisations that we have made in RPIOS. RPiOS will run much faster than Ubuntu in your use case.

RobotsAreCrazy, 2021-03-07:
bit of a waste of having a 8GB pi tho....