raspberrypi-ui/piclone #42: Errors when running Autogen.sh

inadarei, 2022-09-28:
Running on RPI 4 Model B with 8GB RAM

```
❯ hostnamectl
 Static hostname: rpi
      Icon name: computer
      Machine ID: xxx
      Boot ID: xxx
Operating System: Ubuntu 22.04.1 LTS
      Kernel: Linux 5.15.0-1015-raspi
      Architecture: arm64
```

The error itself, after checking out latest code from this repo:

```
❯ ./autogen.sh
+ test -d m4
+ autoreconf -i -f
configure.ac:6: error: AC_CONFIG_MACRO_DIR can only be used once
./lib/autoconf/general.m4:1970: AC_CONFIG_MACRO_DIR is expanded from...
configure.ac:6: the top level
autom4te: error: /usr/bin/m4 failed with exit status: 1
aclocal: error: /usr/bin/autom4te failed with exit status: 1
autoreconf: error: aclocal failed with exit status: 1
```

spl237, 2022-09-28:
This is a known error when running autogen on Ubuntu rather than Debian. The fix is to delete the line AC_CONFIG_MACRO_DIR([m4]) from configure.ac, which is required on Debian.

inadarei, 2022-09-28:
Thank you. That worked like a charm. Any chance we can add this to README, to avoid people re-opening the same issue like I did unintentionally? Happy to submit a PR if that helps.