raspberrypi-ui/piclone #43: Not finding source or destnation drives

ab1jx, 2023-04-28:
This was on Raspbian Buster (after Bullseye was out).  It was a problem with the deb version so I uninstalled  it and built from sources, same thing.  Something similar was found on the Ubuntu site https://askubuntu.com/questions/1081191/backup-failed-error-g-io-error-quark/1465774#1465774  It has to do with g-io-error-quark whatever that is.  Here the lists don't populate and there are the error messages. 
[image]

spl237, 2023-04-28:
Sorry, but as you say on your askubuntu post, this isn't a piclone issue - it is a problem elsewhere on your system, which you are going to need to track down yourself. I've never seen that error, so can't help, I'm afraid.

closed 2023-04-28 by spl237

ab1jx, 2023-04-30:
It's probably not officially supposed to work but I just built pi-clone under Debian Bullseye and cloned the SD card in this Pi 3B, now I'm running the clone (under Debian).  I just followed the directions carefully with regard to installing depends.  It works, the source and destination drive populate correctly, so does making UUIDs.

But I still get these errors, which I guess can be ignored since it worked.  Maybe they're a form of probing, yes, the mount points, that makes sense.
[code]
umount: /dev/sda9: no mount point specified.
umount: /dev/sda8: no mount point specified.
umount: /dev/sda7: no mount point specified.
umount: /dev/sda6: no mount point specified.
umount: /dev/sda5: no mount point specified.
umount: /dev/sda4: no mount point specified.
umount: /dev/sda3: no mount point specified.
umount: /dev/sda2: not mounted.
umount: /dev/sda1: not mounted.
1+0 records in
1+0 records out
512 bytes copied, 0.00333259 s, 154 kB/s
fdisk: write error
mke2fs 1.46.2 (28-Feb-2021)
fdisk: write error
[/code]
I wonder if the raspberrypi retro Buster isn't a little strange, I use it in some cases to get the VC4  support for video streaming.,  mostly I prefer Debian.    It's that Buster where I had the problem with pi-clone.

ab1jx, 2023-05-01:
Solved: This is a booby trap for people who run as root on a Rasberry Pi (under raspbian).  If you use it as the Pi user or even as root under Debian then it works correctly.  I've always run as root and it used to work so something changed.  You're copying an entire OS, it's reasonable to assume you should be root.

spl237, 2023-05-01:
The GIO volume monitor, which is used to detect which drives are available, is accessed via DBus. If you try to access DBus when running as sudo, then you get the system DBus and not the session DBus, and the volume monitor is not available on the session bus. If you look at the desktop file which is provided as part of this project, you will see that it does indeed run piclone as sudo, but crucially does so as "sudo -AE dbus-launch piclone" - the dbus-launch program enables processes run as sudo to access the session DBus.

So there is nothing whatsoever wrong with piclone - the problem is that you were starting it incorrectly from the command line.