raspberrypi-ui/piclone #44: Does piclone (SD Card Copier) work with GPT-partitioned "Hybrid MBR" RPi SSD's? (As is necessary above 2TB!)

holta, 2023-12-06:
Many thanks to RonR who's helped so many people get modern/large SSD drives (larger than 2TB!) working with Raspberry Pi!  Thanks to his lifesaver **mbr2gpt** (interactive bash script) that he's posted here: (as he very actively supports the entire RPi community, amazingly!)

https://forums.raspberrypi.com/viewtopic.php?t=196778

KEY QUESTION: Does anyone know if such GPT-partitioned "Hybrid MBR" Raspberry Pi OS SSD drives can successfully be copied using **Raspberry start menu (top-left of screen) > Accessories > [SD Card Copier](https://forums.raspberrypi.com/viewtopic.php?t=234411#p1433918)** ?

spl237, 2023-12-07:
To be honest, I don't know! SD Card Copier simply duplicates the partition structure (up to a limit of 10 partitions, IIRC) and then does a standard file copy to duplicate everything in each partition. I can't immediately see any reason why it wouldn't work unless there is something peculiar about the gpt partition structure. Have you tried it?

holta, 2023-12-07:
> Have you tried it?

I will ask educators in "wartorn" Cameroon to try it as soon as they can in coming weeks, and report back results.

@spl237 huge thanks for your work & for explaining!

holta, 2023-12-08:
No dice!

Just FYI they (in Cameroon) tried SD Card Copier using the very latest 64-bit Raspberry Pi OS with desktop (from December 5th 2023) and the immediate result is "Non MS-DOS partition table on source." — as seen in this screenshot:

[image]

spl237, 2023-12-08:
OK, thank you for testing.

Without looking into it, I'm not sure if that will be a trivial or a complicated thing to fix. I will put it on the to-do list and have a look at it in the new year.

LinuxfarmerHH, 2025-03-15:
Bought a geekworm with jmicron raid controller and put 2x 5TB drives into it. Tried to copy the system over, with this tool. Result is a non boot and the GPT table from the raid on hdd is marked as damaged.

HDD can be bought as 20TB and SSD with 4TB, so i vote for this feature request.