raspberrypi-ui/piclone #47: Able to compile/build/make piclone on Armbian 23.10.1/Ubuntu 22.04, but no reference to how to create desktop icon, say in Gnome?

Jibun-no-Kage, 2024-02-28:
Able to compile/build/make piclone on Armbian 23.10.1/Ubuntu 22.04, had to add the following:

# apt install intltool libgtk-3-dev

But no reference to how to create desktop icon, say in Gnome?  Application found via search so was easy to add to the Gnome dash panel.

spl237, 2024-02-28:
The icon lives in the package pixflat-icons, so sudo apt install pixflat-icons. But I'm afraid I don't know how to use it in a desktop shortcut on Ubuntu - you'll need to ask someone who knows about Ubuntu for that.

closed 2024-02-28 by spl237

Jibun-no-Kage, 2024-02-29:
Yeah... the icon was one thing.  But the real issue is the application fails once it tries to parse out what is on the source media selected.  Which is odd, given the basic layout of the media is comparable to Pi OS images.  So the fact that the tools fails, means there is something 'Pi' specific in play.  To bad, the tool is quite nice, would be really cool if it was more cross-platform, at least in the same distribution family of debian/ubuntu for example.